minwoo0611/MA-LIO #1: Do you have the arxiv version of the paper?

Gatsby23, 2023-04-20:


minwoo0611, 2023-04-20:
The specified version has not yet been published on arXiv. We have recently completed the review process, and we anticipate being able to upload the preprint within the next month. Thank you for your inquiry.

Sincerely,
Minwoo Jung

closed 2023-04-20 by minwoo0611

Gatsby23, 2023-04-28:
> The specified version has not yet been published on arXiv. We have recently completed the review process, and we anticipate being able to upload the preprint within the next month. Thank you for your inquiry.
> 
> Sincerely, Minwoo Jung

Question about the Point-wise. Does it mean using the point-to-point constraint? It means you use the GICP to update the state instead of the original ICP in the FAST-LIO2？

minwoo0611, 2023-05-01:
In our paper, point-wise means assigning a unique uncertainty to each point. This uncertainty is used in the error state extended Kalman filter, and the measurement model is the same as in FAST-LIO2. In other words, we apply the point-to-plane model used in FAST-LIO2, but you can think of it as an added weight model according to the uncertainty.

Gatsby23, 2023-05-01:
Thank you for your patient guidance. I hope that your paper will be accepted by the reviewers as soon as possible.