minwoo0611/MA-LIO #2: hi~great work! Do you have the specific time for opening the source code? Thank you!

lhthq, 2023-05-29:
hi~great work! Do you have the specific time for opening the source code? Thank you!

minwoo0611, 2023-05-30:
Thank you for showing interest in my code. 

I am eager to release it as soon as possible, but I am currently working on generalizing the code. I can't say for sure how long this will take, as it is being delayed due to my personal projects and other schedules. 

However, I plan to make it public by **June** at the latest. If my code needs to be used in papers or projects for comparison with other methodologies, I think I could provide it tailored to specific targets. Thank you once again for your interest, and I will try to proceed as quickly as possible.

minwoo0611, 2023-06-25:
@Ihthq

Finally, we've released our code! 
Given this, I'll be closing this issue now.
Thank you for your patience.

closed 2023-06-25 by minwoo0611