minwoo0611/MA-LIO #3: fail

HIT-Ygq, 2023-06-28:
Very good work! I can run` roslaunch ma_lio mapping_city.launch` successfully, but `roslaunch ma_lio mapping_urban.launch `will fail, have you encountered this?
[image]

minwoo0611, 2023-06-28:
It appears that your issue may have been resolved, but I've added some additional comments to further clarify things.

The launch file I utilize for urbanNav is designed for a 2 LiDAR system, whereas for the city dataset, it is tailored for a 3 LiDAR system. As detailed in the ReadMe, should you wish to operate MA-LIO with a different number of LiDARs, certain portions of the code need to be altered in accordance with the instructions provided in the ReadMe.

I appreciate your efforts in testing MA-LIO. Should you have any additional questions or concerns, please feel at ease to raise them in the issue tab.

HIT-Ygq, 2023-06-28:
Thanks, I'm testing a single LiDAR.  Would you be able to write a detailed description?

minwoo0611, 2023-06-28:
If your intention is to use a single LiDAR system, modifications need to be made in both the yaml file and the code. Here are the three key changes:

1. Modify the yaml file. You should adjust each variable, as outlined in the ReadMe, to be a 1 size array with a single element.
2. Update the `laserMapping.cpp` file. The `ApproximateTime`, `Synchronizer`, and `cbk` function should all be modified. Instructions on how to adapt the code for a single LiDAR in the MA-LIO can be found within the function. To find this information easily, you can search for the word "Especially" within `laserMapping.cpp`.
3. Lastly, the `use-ikfom.hpp` needs to be adjusted. The extrinsic variable should match the number of your LiDARs, in this case, 1.

I sincerely hope that these instructions will be beneficial to your project. Thank you for your ongoing engagement.

HIT-Ygq, 2023-06-28:
I modified the following points, but the compilation error, can you provide comments?
[image]
[image]
[image]
[image]

hr2894235132, 2023-12-05:
> I modified the following points, but the compilation error, can you provide comments? ![101](https://private-user-images.githubusercontent.com/90457565/249379168-aecfa445-75e3-4c82-acfd-43594f687a00.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3NDA5ODAsIm5iZiI6MTcwMTc0MDY4MCwicGF0aCI6Ii85MDQ1NzU2NS8yNDkzNzkxNjgtYWVjZmE0NDUtNzVlMy00YzgyLWFjZmQtNDM1OTRmNjg3YTAwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA1VDAxNDQ0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdiNTBkY2NjZWEzNTE1M2VhY2QyZjBlNzJmMGEyNDY1NjZiZDkxNDcwOTFlNGVjNTYxNmEzODYzOTUxMDNhOTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.FNPM3UUKXPkSW3ZawFHhpl4CWpEqEeBKGsMY1Co77kg) ![102](https://private-user-images.githubusercontent.com/90457565/249379190-5a0ccd61-c7ec-4f0d-b363-58f6fc417949.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3NDA5ODAsIm5iZiI6MTcwMTc0MDY4MCwicGF0aCI6Ii85MDQ1NzU2NS8yNDkzNzkxOTAtNWEwY2NkNjEtYzdlYy00ZjBkLWIzNjMtNThmNmZjNDE3OTQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA1VDAxNDQ0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNiYmI3MWYyN2Q0ZTIxZDAwYWIxNzI5Njc3Zjc1OGE5MzNjMzQzOWU0OTc3MjljMTAzODE5OGJmNTIxYTA3NTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.P7oZ9PQY8ujeCJgAsP_I5us9X73csHlKahAqiebJTsM) ![103](https://private-user-images.githubusercontent.com/90457565/249379212-1f2f1b93-c6cc-47ce-998d-c4f7b4229ef2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3NDA5ODAsIm5iZiI6MTcwMTc0MDY4MCwicGF0aCI6Ii85MDQ1NzU2NS8yNDkzNzkyMTItMWYyZjFiOTMtYzZjYy00N2NlLTk5OGQtYzRmN2I0MjI5ZWYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA1VDAxNDQ0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzZDNhNzYyZGEyNzIwNWNiMTliMDA4NjgxOTU3MTAzYjUyN2VlNTc1MTdlYjcwNjMzYWQ0ZWI5MjViMTJkNjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.epV0jqWh4zI8GaK0h8hcsDafgxNPp2egrtxFBnXvUxg) ![104](https://private-user-images.githubusercontent.com/90457565/249379230-65863e23-2641-459c-b1fd-abd3ee7fe1b3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3NDA5ODAsIm5iZiI6MTcwMTc0MDY4MCwicGF0aCI6Ii85MDQ1NzU2NS8yNDkzNzkyMzAtNjU4NjNlMjMtMjY0MS00NTljLWIxZmQtYWJkM2VlN2ZlMWIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA1VDAxNDQ0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJiMzg4YjMwNmM1ZGFjMjdhMTEyNWEzODgzMmM2MjUxMDQ0OTg1NjM2MjUzNTc3YWNkODMwM2RlYWJhYTdhZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.3y_RLrSWMJR0Q1HADe8lyYtDYwUxxT6kruFENHpOuEg)

Hello, how did you test a single lidar?

minwoo0611, 2023-12-05:
@hr2894235132 
Hello HaoRui, can you give me a more information about your problem?

hr2894235132, 2023-12-05:
> @hr2894235132 Hello HaoRui, can you give me a more information about your problem?

Thank you for your prompt reply! First of all, thank you very much for your excellent work for me to learn. in fact, I have been doing fast-lio and voxel_Map related tests before, and I saw that the friend above tested the MA-LIO single lidar, so I want to understand the effect of the MA-LIO single lidar first.

minwoo0611, 2023-12-05:
You're welcome! It's great to hear about your interest and experiments with fast-lio and voxel_Map. Regarding the use of MA-LIO with a single LiDAR, here are some key points to consider:

1. **Undistortion of Points**: MA-LIO enhances point undistortion by utilizing a continuous trajectory estimator. This approach provides a smoother outcome compared to traditional linear time interpolation, leading to more accurate point undistortion.

2. **Residual Weight Adjustment**: In environments where the signal might be degenerate, MA-LIO adjusts the residual weight between the LiDAR and IMU based on localization weight, which could be beneficial even for a single LiDAR setup.

3. **Lidar Uncertainty**: The concept of LiDAR uncertainty, a feature of MA-LIO, is also applicable to single LiDAR systems. This means each measurement can have a varying weight, enhancing the accuracy of the data.

It's important to note that while MA-LIO is primarily designed for asynchronous multiple LiDAR integration, these features can still be effectively applied to a single LiDAR setup, although the overall impact might be less pronounced compared to a multi-LiDAR configuration.

closed 2023-12-07 by minwoo0611