minwoo0611/MA-LIO #4: Seeking Suggestions on Adjusting Parameters of Uncertainty

thisparticle, 2023-07-02:
Hello, I would like to ask if you have any suggestions for adjusting the parameters of uncertainty. 

I tried to test with your code in a long tunnel with data from a single avia solid-state lidar, and I found that the lidar poses diverge very quickly in this case. I tried to improve the robustness by adjusting the parameters of uncertainty, but no success so far. 

Another small problem I have is that with only lidar and imu data, the positioning is easily affected by environmental degradation, especially with small FOV lidar like Avia. At this point, your algorithm solves the degradation problem by reducing the weight of the lidar, but it tends to be too dependent on the imu results. And the IMU results usually diverge very quickly. I want to ask you how your solved this problem and got good results in the City02 tunnel experiment.

minwoo0611, 2023-07-02:
Dear @thisparticle,

Firstly, I thank you for your interest in MA-LIO. As you have pointed out, conducting SLAM in tunnel environments is indeed challenging due to errors arising from the measurement model. More specifically, the extensive tunnel environment that you mentioned presents even greater difficulties. In the case of our City02 tunnel, which is approximately 350m in length, we didn't encounter significant issues while running MA-LIO. This was primarily due to the tunnel's moderately wide interior, which allowed for sufficient scan-to-map matching. It's speculated that the matching correspondence errors did not amplify the IMU bias in the state model to the point of failure.

Our uncertainty method and localization weight do improve accuracy in degenerate environments, but this assumes our state successfully infers velocity and IMU bias. When we use localization weight to rely on the IMU model, if the accuracy of the IMU model itself is substandard, state estimation becomes difficult. Thus, if the correspondence error in the tunnel persistently and significantly deviates the IMU model, it can pose a problem that cannot be resolved merely through parameter adjustments. In fact, our MA-LIO also failed to map the Tunnel-1 dataset from UrbanNav, which is a tunnel approximately 1.72km in length.

In particular, Avia, the LIDAR you are using, presents individual scans that are notably sparse, which could exacerbate the problem in tunnel environments. In our city dataset, we utilized Livox Avia and Livox Tele, but we essentially had Ouster, a dense LIDAR.

Let me reiterate that regardless of the algorithm, environments like tunnels or other narrow spaces are extremely challenging, especially if a narrow-view LIDAR is used. The task of mapping becomes exceedingly tough. However, if one needs to map tunnel environments, utilizing additional values such as intensity to find accurate mapping correspondences can be crucial.

I hope this response addresses your queries, @thisparticle. I have also attempted, albeit unsuccessfully, to solve the UrbanNav tunnel dataset through various algorithms. This perspective shows that there are still areas in SLAM research that require ongoing exploration. Once again, I wish you success in resolving the problem at hand.

Best reagrds,
Minwoo

thisparticle, 2023-07-03:
> Dear @thisparticle,
>
> Firstly, I thank you for your interest in MA-LIO. As you have pointed out, conducting SLAM in tunnel environments is indeed challenging due to errors arising from the [...]

Thank you for your prompt response and valuable insights. I truly appreciate your assistance.

closed 2023-07-03 by minwoo0611