minwoo0611/MA-LIO #5: Running problems on public datasets

baibai09, 2023-07-02:
Very good work！ 
When I use a single LIDAR, it always crashes. I have tried various solutions to this problem, but none of them have been successful. As a result, I am turning to you for help.

1. LIO-SAM Dataset [walk](https://drive.google.com/drive/folders/1gJHwfdHCRdjP7vuT556pv8atqrCJPbUq)
  The map will drift and the program will crash.
[image]
[image]

2. utbm dataset [utbm](https://epan-utbm.github.io/utbm_robocar_dataset/#Downloads)
The program just crashes
[image]

minwoo0611, 2023-07-02:
Dear @baibai09,

Firstly, thank you for testing MA-LIO.
I find that a bit more information would greatly assist me in fully understanding and addressing the issue at hand.

Could you elaborate on how you have modified the code to accommodate different datasets? 

Resolving the issue using solely the image you shared presents a significant challenge. Additional related materials, such as snippets of your code change, would be highly beneficial in facilitating a more efficient and accurate troubleshooting process.

Additionally, I have not yet executed tests using the UTBM dataset. Nonetheless, the LIO-SAM walking dataset is functioning without any issues when interfaced with the MA-LIO system.

[image]

[image]

baibai09, 2023-07-03:
Thank you very much for your reply!

Here are some of my configuration changes and the code：
Is it because I'm using Ubuntu 18.04, not Ubuntu 20.04?

[image]
[image]
[image]
[image]

minwoo0611, 2023-07-03:
Thank you for sharing your code!

The things you change are all correct!
However, the extrinsic between LiDAR and IMU seems be false.

When we see the config.yaml of LIO-SAM, the extrinsic between two sensors is  [-1, 0, 0, 0, 1, 0, 0, 0, -1].
In LIO-SAM, extrinsic calibration is defined as IMU to LiDAR; however, our definitions is LiDAR to IMU (In fact, it does not a problem in the problem as the inverse of extrinsic is sams) with the form of quaterion [qw, qx, qy, qz].
Therefore, the extrinsic_R in yaml file should be changed as [0, 0, 1, 0].

I hope that this answer can solve you problem.

baibai09, 2023-07-03:
Thank you for your assistance. Unfortunately, the issue persists even after I changed the external reference. I will continue to investigate the problem. 
Thank you again for your response.

closed 2023-07-03 by baibai09