minwoo0611/MA-LIO #10: How to disable extrinsic estimation on translation but keep the estimation on rotation?

zzxxtt, 2023-07-28:
Hello, thank you for the great work. 
When using ma-lio I found one problem, the extrinsic estimation on translation is not so accurate. translation can different from groundtruth CAD for 5cm or more. But I still want to estimation rotation extrinsic for small install error. 
Anyways to do this?

minwoo0611, 2023-07-29:
@zzxxtt 

Thank you for your interest and inquiry about MA-LIO. 
Theoretically, there is nothing wrong with the calibration method, but it is true that there are some translation inaccuracies for some systems when used in practice. 

Anyway, if you want to use extrinsic calibration only for the Rotation part as you like, you can set the Jacobian for translation to 0.

It will be solved by [modifying the part](https://github.com/minwoo0611/MA-LIO/blob/f0ee6b31ee9c9f4bc684c3e419133e4bb3548a0b/MA_LIO/src/laserMapping.cpp#L693) in as below. 
From
`ekfom_data.h_x.block<1, 3>(i, 6 + 3 * (lid_num + lid_idx)) << VEC_FROM_ARRAY(C); `
to
`ekfom_data.h_x.block<1, 3>(i, 6 + 3 * (lid_num + lid_idx)) << 0, 0, 0;
`

If you need any additional help, please feel free to ask.

closed 2023-07-29 by minwoo0611

zzxxtt, 2023-07-30:
It works, thank you very much