minwoo0611/MA-LIO #13: LIO-SAM Park dataset not working

saran1504, 2023-09-26:
Hi. Good work with the mapping. Everything works fine and I tried with LIO-SAM walking dataset and it works really good. I am testing your system as a part of my thesis and would like to work with the dataset with GPS data. LIO-SAM parking dataset has the GPS data and I am trying to make your system work with that dataset. But unfortunately it is throwing error after few seconds. Could you please help me with this regard? I am not sure how to put the transformations and rotations matrix for the park dataset.

minwoo0611, 2023-09-27:
Hello @saran1504,

Thanks for diving into our system with the LIO-SAM datasets. It's great to hear that the walking dataset is running smoothly for you.

Regarding the parking dataset that incorporates GPS data, I'm here to help. In order to provide the most accurate assistance, I'd appreciate some more information. Could you share more details about the error messages you've encountered and any adjustments or configurations you've made? With that information in hand, I'll be better equipped to guide you through potential solutions.

Awaiting your response, and thanks for your patience!

saran1504, 2023-09-27:
Thanks for your reply. Since park dataset is same as walking with single lidar system I didn't change anything in the code. I just tried to change in the parameters file. I tried to enter the extrinsic T and R values from liosam one. But all I get is a dense point cloud and a curved one. A suggestion was given previously that liosam is lidar to imu and Malio was from imu to lidar and to change 0,1,0 to 0,0,1. I tried that and then the process dies after 70 seconds . I need help with the yaml file to calculate the extrinsic_T and R values. Or is there anything else to change more for the park dataset?

saran1504, 2023-09-27:
Walking dataset is working perfectly and I might not need any help in that. I just need help with the park dataset to work. I will try to look into the issue you mentioned and thank you for your prompt response. Good luck with your conference and I will update here if I have some improvement.

minwoo0611, 2023-09-27:
I'll also upload the results it when I come back to the office.

Thank you for your interest in our project!

The issue will be closed for a while, and will be re-opened if there are additional updates.

minwoo0611, 2023-10-10:
Hello @saran1504,

As promised, I returned to my office and attempted to launch MA-LIO for the park dataset. Based on my findings, everything seems to be working as expected. Attached are the images from LIO-SAM (top) and MA-LIO (bottom).

[image]

[image]

I adhered to the instructions I previously provided, so I'm not sure why it's not functioning in your setup.

If you require the trajectory output from MA-LIO, I can provide it in TUM format.

Thank you.

saran1504, 2023-10-10:
Hi,

Thanks for helping me out. Could you provide me the .yaml file for which
the park dataset worked? That might help me.

Thanks,
Saran

minwoo0611, 2023-10-10:
common:
    lid_topic: ["/points_raw"]
    lid_type: [2]           #1-Livox 2-Velodyne 3-Ouster
    N_SCANS: [16]
    point_filter_num: [1]
    lid_num: 1
    imu_topic: "/imu_correct"
    time_sync_en: false           # ONLY turn on when external time synchronization is really not possible
    time_offset_lidar_to_imu: 0.0 # Time offset between lidar and IMU calibrated by other algorithms, e.g. LI-Init (can be found in README).
                                  # This param will take effect no matter what time_sync_en is. So if the time offset is not known exactly, please set as 0.0

preprocess:
    timestamp_unit: 0           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 0

mapping:
    acc_cov: 0.011197412605492375
    gyr_cov: 0.010270904839480961
    b_acc_cov: 0.00011751767903346351
    b_gyr_cov: 0.000091355383994881894
    fov_degree:    180
    det_range:     100.0
    extrinsic_est_en: false      # true: enable the online estimation of IMU-LiDAR extrinsic,
    extrinsic_T: [0, 0, 0]
    extrinsic_R: [1, 0, 0, 0]
    
    # extrinsic_T: [-0.0, 0.3105, 0.2068, -0.5502, -0.0471, 0.2925 (Ouster),  -1.5060, 0.017, 0.024(Velo), 0.0719, -0.4006, 0.0685(Livox)]
    # extrinsic_R: [0.9997, 0.0007, -0.0108, -0.0232, 0.9999,   -0.0028,   -0.0028,   -0.0108, 1, 0, 0, 0, 0.9998,   -0.0115,    0.0023,   -0.0188]

publish:
    path_en:  true
    scan_publish_en:  true       # false: close all the point cloud output
    dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: 300                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
uncertainty:
    point_cov_max: 0.00125
    point_cov_min: 0.00075
    plane_cov_max: 1        # 1/2 = 1
    plane_cov_min: 0.8      # 1/0.8 = 1.25
    localize_cov_max: 2    
    localize_cov_min: 0.3  
    localize_thresh_max: 0.7 #upper, max
    localize_thresh_min: 0.2 #lower, min
cov_threshold: 1.0
Hi Saran,

Certainly, here's the .yaml file that I used for the park dataset. If you're still encountering issues after reviewing this file, I'll create a separate branch to upload the version that worked for me. This should aid you in resolving any discrepancies.

Best,
Minwoo

saran1504, 2023-10-10:
Hi Minwoo,

The above mentioned parameters worked for me too. There was a small
variation in the extrinsic_t which was making my program crash. Thank you
so much for helping me out. All the best for your future regards.

Thanks,
Saran
ᐧ

closed 2023-10-10 by minwoo0611