minwoo0611/MA-LIO #17: compoundPoseWithCov()  seems to be ineffective

Liansheng-Wang, 2023-12-11:
In the code laserMapping. cpp, it seems that the functions compoundPoseWithCov() and compoundInvPoseWithCov() have not updated any cov.

minwoo0611, 2023-12-11:
Hello @Liansheng-Wang,

Thank you for reaching out with your question regarding the functions `compoundPoseWithCov()` and `compoundInvPoseWithCov()` in `laserMapping.cpp`. I appreciate your attention to the details in the code.

From your query, it seems you are referring to these specific lines: 
https://github.com/minwoo0611/MA-LIO/blob/b911ad69c3a8bf43133333e8913d0d42cb5dabba/MA_LIO/src/laserMapping.cpp#L1040-L1042

In the original code, the line was:
```cpp
pose_unc[num].push_back(kf.lidar_uncertainty[num][i]);
```

You've correctly identified an issue. The intention was to update the `pose_unc[num]` with the calculated `pose_point` resulting from the `compoundPoseWithCov()` and `compoundInvPoseWithCov()` functions. However, the initial implementation mistakenly did not update the covariance values as expected. 

I've now revised the code to correctly save the `pose_point`, including its covariance, into `pose_unc[num]`. This should resolve the issue you observed where the covariance wasn't being updated as it should have been.

If your question pertains to a different aspect or if there's anything else unclear about the calculation logic, please feel free to add further comments. Again, thank you for pointing out this issue!

Liansheng-Wang, 2023-12-11:
Thank you for your reply, and also for the project you have open-sourced.

closed 2023-12-11 by minwoo0611