minwoo0611/MA-LIO #22: About the multi-livox Lidar calibration method

hr2894235132, 2024-03-12:
First of all, thank you for your excellent work, we are ready to use two front-to-back placed avias for multi-scenario testing.Do you have any recommended calibration algorithms that can calibrate this kind of scene without common view?

minwoo0611, 2024-03-12:
Hello @hr2894235132 

I recommend you to use this repository, [mlcc](https://github.com/hku-mars/mlcc)

I hope this calibration method can resolve your problem.

hr2894235132, 2024-03-13:
> Hello @hr2894235132
> 
> I recommend you to use this repository, [mlcc](https://github.com/hku-mars/mlcc)
> 
> I hope this calibration method can resolve your problem.

thanks for your reply! I'll record the dataset today and test mlcc,

closed 2024-03-13 by minwoo0611

hr2894235132, 2024-03-27:
Hello, when I was testing with my two avias, and then changed the topic order of the two lidars before and after, the results were inconsistent, and when I changed the backward lidar to the first one, there was a shift.
Which configuration file should be used for the Hilti dataset? I'd like to use a public dataset to do the above test and see if it's a problem with my dataset.