minwoo0611/MA-LIO #23: About Lidar switching

hr2894235132, 2024-03-27:
Hello, I tested with two avias, and at the beginning there was a small oscillation at the end, and after changing the corresponding parameters according to the discussion in issuse, I successfully completed the handheld test.
But in one scene there was an error problem due to radar switching, when I was testing with my two avias, and then changed the topic order of the two lidars before and after, the results were inconsistent, and when I changed the backward lidar to the first one, there was a shift.Which configuration file should be used for the Hilti dataset? I'd like to use a public dataset to do the above test and see if it's a problem with my dataset.
By the way, have you ever tried the performance of the ma-lio at high speeds.

minwoo0611, 2024-03-27:
Dear @hr2894235132, 

Thank you for reaching out and sharing your experiences and issues encountered while testing with two avias. Regarding the small oscillation observed initially, it's good to hear that adjusting parameters based on our previous discussions helped you overcome this challenge and successfully complete the handheld test.

In terms of the error encountered due to radar switching and the inconsistency observed when changing the topic order of the two lidars, this is indeed a peculiar situation. Unfortunately, our project concluded a year ago, and we do not have specific configuration parameters tailored for the Hilti dataset. However, as we outlined in our paper, the parameters we selected show only slight variations from those discussed. We encourage you to use these guidelines when testing with the Hilti dataset, referencing other configuration files as necessary.

It's worth noting that MA-LIO utilizes synchronization, designed to adjust the primary and secondary lidar based on their timestamps (every iterations). Therefore, the results should remain consistent regardless of which lidar is designated as primary or secondary. A potential area for investigation could be to check, during each iteration, which LiDAR scan combination is being used, based on their scan timestamps.

Regarding your inquiry about the performance of MA-LIO at high speeds, such as in UAV applications, we have not conducted tests under these conditions. Given that MA-LIO employs B-Spline for data smoothing, we hypothesize that its performance may be compromised at high speeds. This is because B-Spline tends to smooth out the data, potentially leading to inaccuracies in instances where IMU values peak, and instantaneous accelerations or angular velocities are high.

We hope this information proves helpful for your ongoing tests and research. Please feel free to reach out if you have further questions or need additional clarification.

Best regards,
Minwoo

hr2894235132, 2024-03-27:
Dear @minwoo0611 ,
First of all, thank you for your prompt, detailed and helpful reply. 
For the testing of the Hiliti dataset, I will follow your suggestion to do some sequence testing. And for my own platform, I will also do some checks and verifications as you mentioned.
Later, I will put the handheld device on the drone for testing, hoping to help our drone use multiple lidar positioning problems.
Finally, I would like to thank you again and hope to maintain friendly communication in the future.
Best wishes!

closed 2024-03-27 by minwoo0611