minwoo0611/MA-LIO #33: Build isues in Ubuntu 20.04 and noetic

pk30123, 2024-09-03:
Hi everyone, 
I face the following errors when i catkin build in both ma_lio and file player. Any ideas will help.

Errors     << ma_lio:make /home/malio/logs/ma_lio/build.make.002.log                                                                                                                                       
In file included from /usr/include/boost/preprocessor/tuple/elem.hpp:23,
                 from /usr/include/boost/preprocessor/arithmetic/add.hpp:21,
                 from /usr/include/boost/mpl/aux_/preprocessor/def_params_tail.hpp:66,
                 from /usr/include/boost/mpl/aux_/na_spec.hpp:28,
                 from /usr/include/boost/mpl/next_prior.hpp:18,
                 from /usr/include/boost/mpl/next.hpp:17,
                 from /usr/include/boost/mpl/list/aux_/push_front.hpp:18,
                 from /usr/include/boost/mpl/list/list0.hpp:19,
                 from /usr/include/boost/mpl/list/list10.hpp:18,
                 from /usr/include/boost/mpl/list/list20.hpp:18,
                 from /usr/include/boost/mpl/list.hpp:36,
                 from /usr/include/boost/math/policies/policy.hpp:9,
                 from /usr/include/boost/math/policies/error_handling.hpp:21,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/noetic/include/ros/time.h:58,
                 from /opt/ros/noetic/include/ros/ros.h:38,
                 from /home/malio/src/MA-LIO/MA_LIO/src/laserMapping.cpp:11:
/home/malio/src/MA-LIO/MA_LIO/src/preprocess.h:33:6: warning: ‘using uint16_t = uint16_t’ is deprecated: use std::uint16_t instead of pcl::uint16_t [-Wdeprecated-declarations]
   33 |     (uint16_t, ring, ring)
      |      ^~~~~~~~
In file included from /usr/include/pcl-1.10/pcl/PCLPointField.h:10,
                 from /usr/include/pcl-1.10/pcl/conversions.h:46,
                 from /opt/ros/noetic/include/pcl_conversions/pcl_conversions.h:44,
                 from /home/malio/src/MA-LIO/MA_LIO/src/laserMapping.cpp:18:
/usr/include/pcl-1.10/pcl/pcl_macros.h:94:9: note: declared here
   94 |   using uint16_t [[deprecated("use std::uint16_t instead of pcl::uint16_t")]] = std::uint16_t;
      |         ^~~~~~~~
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/malio_mapping.dir/build.make:63: CMakeFiles/malio_mapping.dir/src/laserMapping.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2626: CMakeFiles/malio_mapping.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/malio/build/ma_lio; catkin build --get-env ma_lio | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

TIA

minwoo0611, 2024-09-04:
Hello, @pk30123,

Thank you for using MA-LIO. I'm not entirely sure why you're encountering this error on your PC. I recently re-cloned the repository and did not face any issues with `catkin build`.

After researching the error "c++: fatal error: Killed signal terminated program cc1plus," I found some cases similar to yours. You might find this GitHub discussion helpful: [MoveIt2 Issue #1276](https://github.com/moveit/moveit2/issues/1276). It suggests that the problem could be due to your PC running out of RAM during the build process.

Could you please check your RAM usage to see if that's the case?

Best regards,  
Minwoo

pk30123, 2024-09-05:
Hello Minwoo,

I think RAM was the problem, I do not get it when building sequentially. Buit i get this new error.

Errors     << ma_lio:make /home/malio/logs/ma_lio/build.make.001.log
In file included from /home/malio/src/MA-LIO/MA_LIO/include/associate_uct.hpp:5,
                 from /home/malio/src/MA-LIO/MA_LIO/src/laserMapping.cpp:32:
/home/malio/src/MA-LIO/MA_LIO/include/common_lib.h:8:10: fatal error: ma_lio/Pose6D.h: No such file or directory
    8 | #include <ma_lio/Pose6D.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/malio_mapping.dir/build.make:63: CMakeFiles/malio_mapping.dir/src/laserMapping.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2626: CMakeFiles/malio_mapping.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/malio/build/ma_lio; catkin build --get-env ma_lio | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

TIA,
Pradheep

pk30123, 2024-09-06:
Hi Minwoo,

Thanks for the quick reply . RAM update works. but i do not have a idea why another error occurs.
process[laserMapping-2]: started with pid [91]
Multi thread started 
~~~~/home/malio/src/MA-LIO/MA_LIO/ doesn't exist

TIA,

pk30123, 2024-09-19:
Thanks it was solved because of RAM it seems