minwoo0611/MA-LIO #34: Empty RVIZ

pk30123, 2024-09-25:
Hi team,
i tried MA-LIO in my databut the RVIZ is empty. The idea is similair to urbannav with a ouster and a velo. What parameter do i check?
[image]
TIA,

minwoo0611, 2024-09-26:
Hello, @pk30123 

Thank you for using MA-LIO. 
I’m not entirely sure what the issue is based on just the RViz view. 
Could you please check the terminal output for more details?

Thank you

pk30123, 2024-09-26:
with my dataset 

auto-starting new master
process[master]: started with pid [97]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to beaf3f7e-7be9-11ef-8c0f-5c879c224638
process[rosout-1]: started with pid [107]
started core service [/rosout]
process[laserMapping-2]: started with pid [110]
process[rviz-3]: started with pid [115]
Multi thread started 
~~~~/home/malio/src/MA-LIO/MA_LIO/ doesn't exist

in urbannav i get the output 

**** Multiple Asynchronous LiDAR Inertial Odometry (MA-LIO) ****

[Timestamp of Current State]: 0 secs
[Position]: 
	[x]: -0.40022 meter
	[y]: 14.761 meter
	[z]: -0.25905 meter
[Orientation]: 
	[qx]: -0.0022392
	[qy]: -0.0031923
	[qz]: 0.023296
	[qw]: 0.99972
[Velocity]: 
	[x]: -0.029662 m/s
	[y]: 5.4306 m/s
	[z]: -0.093548 m/s
[LiDAR-IMU Extrinsic 0]: 
	Timestamp for the Lastest Point: 1621322436.2058365345 secs
	Translation: 
		[x]: 0.30135 meter
		[y]: -0.014685 meter
		[z]: 0.088213 meter
	Rotation: 
		[qx]: 0.0011741
		[qy]: 0.46298
		[qz]: -0.018718
		[qw]: 0.88617
[LiDAR-IMU Extrinsic 1]: 
	Timestamp for the Lastest Point: 1621322436.1959002018 secs
	Translation: 
		[x]: 0.01787 meter
		[y]: 0.0090866 meter
		[z]: 0.27047 meter
	Rotation: 
		[qx]: 0.0018756
		[qy]: -0.0021802
		[qz]: -0.00050429
		[qw]: 1
[Temporal discrepancy between LiDAR 0 and 1]: 9.9363 msecs
[Total Distance]: 15.562 meter
[Duration]: 7.7168 secs
[Map Point Count (Downsample w/ 0.5)]: 14264
[Computation Time]: 49.47 msecs
[Computation Time (Avg)]: 42.533 msecs
[RAM Allocation]: 241.54 MB
[VSZ]: 848.78 MB

pk30123, 2024-09-26:
[laserMapping-2] process has died [pid 14000, exit code -11, cmd /home/malio/devel/lib/ma_lio/malio_mapping __name:=laserMapping __log:=/root/.ros/log/ce8756c2-7bfb-11ef-bad5-5c879c224638/laserMapping-2.log].
log file: /root/.ros/log/ce8756c2-7bfb-11ef-bad5-5c879c224638/laserMapping-2*.log

This error is also there on rebuilding

pk30123, 2024-09-27:
Hi, is there any parameter to change icp characteristics so even less common space is needed ? Also how to select the point filter num parameter ,is there a rule of thumb or any?

TIA

minwoo0611, 2025-05-12:
To reduce the number of parameters, we adjusted the filter number parameter to 4–8 to accommodate varying point cloud densities, depending on the available computing resources.

closed 2025-05-12 by minwoo0611